mrisc32/mrisc32 #95: Roadmap ...

gschadow, 2019-10-20:
This is an exciting project, I come here with a dual interest: hobbyist just trying to learn how to build a CPU, but then also still considering my real work and how improved hardware might help. You are obviously way ahead of what I with my limited time that I should spend on a hobby might be able to get to by myself, but I love your way of thinking. So now I wonder where could this project go? How much of an impact could it have in the real world? 

So here are a long set of brainstorming questions:

Let's say you are done (and your seem to be pretty close to done) what then? 
What might be the next steps from the CPU to an actual instance of a computer that people could use? 
Is this ever going to get anywhere, and what would it take to get there? 
Is there ever a chance to actually compete with the architectures that market provides? 
On performance and price? And what might it take? Angel VC Investment? Open Source Community support? 
I see you have been on this project since end of 2017 and you are still committing, albeit less. Would you run out of steam? 
How do you see yourself comparing with the ForwardCom project? And how do you see that project going? 
What is the role of custom hardware in an age where most hardware is being virtualized in "the cloud"? (I used to rent a cage in a data center, but 4 years ago I dissolved all of this and now do everything on AWS. There was an interesting cloud computing provider ScaleWay, who used their own custom hardware, but they had severe limitations that I don't think anyone can run production on them, still, it's an example for the role of custom hardware today.)
How might your project revolutionize the world? Or will it only ever be a thing of a 1-off hobbyist?
I see you already ported GNU binutils, that's cool. What would be the operating system? Linux? NetBSD? (NetBSD has good track record of support of diverse hardware old and new) Minix? Or maybe no UNIX at all? 
Maybe its use is a single user computer like the LISP Machine? (The point being: with cheap hardware, why must everything be thought about as a multi-user system? In the end, what any one operation cares about is run their one WebAPP, their one database, the multi-user multi-processing environment seems like legacy, unnecessary if you can throw hardware at it! Definitely worth considering not running generic context switches in the middle of your parallel vector pipeline operation.)
Might something be gained by setting up a foundation and attract tax exempt money? (I could help with that and even spend something. It's probably too early.)

In the rest here I put my reaction on your blog post to demo my excitement and my personal interest in the application of this. 

--------------------------
I love this!

Someone was asking about string processing, and I agree that it’s one of the most important bread and butter tasks. I think your handling of packed data types is very cool! I actually found your page by asking about “string processing on 32-bit architectures” as I am inspired by Ben Eater’s breadboard CPU project on YouTube wondering how my ideal CPU would be designed.

So your way of dealing with packed data types (90% of strings in UTF-8 will be one byte per character) is already a 4 times speed-up by processing 4 characters in the same number of cycles that on wasteful UTF-16 or -32 representation (e.g. Java). Now I am considering throwing your vector registers at this problem! With your 32 x 32 vector registers, you would get to process strings in 128 character chunks.

Another use of this is bitmap processing with RGBA values that also fit neatly into 32 bits and you could add or multiply each byte individually.

Now bring this down to a nibble (4-bit) and implement BCD arithmetic? I guess BCD fell out of fashion somewhere in the 1980s, but in financial and database applications it would seem to be still a common thing, considering Oracle INTEGER and DECIMAL types and PostgreSQL’s bigint and big decimal. Java has those too.

Or maybe, instead of complicating things with BCD one might combine multiple registers of these vector fields to add with carry across more than one vector component? That way you could add 64 and 128 or higher integers. This might be useful for cryptographic operations like SHA-256, SHA-512, where your would combine 8 or 16 components in your 32×32 vectors.

Which brings me to another consideration. Graphics, ray tracing 3D, and video processing stuff are all well and good (though beyond both my expertise and interest), but speeding up data and symbolic processing is my interest. So, that packed data types go a long way already. Now what about more considerations of parallel processing pipelines. Thinking database join operations, and parallel execution of declarative and functional programming languages. What processor features would make a huge performance gain there? I’m still brainstorming about this …

So what is your road-map? Your CPU will be implemented on an FPGA or is there nowadays a way to produce custom chips?

mbitsnbites, 2019-10-21:
😁 I'm glad that my project can be a source of inspiration to others. I'll read through your questions and suggestions.

mbitsnbites, 2019-10-21:
A few quick answers... Since (so far) this is pretty much a solo project, and an ambitious one, I have to constantly priorotize and move my attention to parts that has the most impact at any given time.

I have worked on the following things, up to a point where I can see results:

* The MRISC32 ISA design.
* A [port of binutils](

gschadow, 2019-10-21:
Fair enough. I had asked the same questions to Agner Fog with his [ForwardCom](https://www.forwardcom.info/) architecture. I'm interested to know if you had seen it and what you think the biggest differences are in your understanding.

Agner Fog is also saying that the bottleneck for any progress is people. But you seem further along as he is, since you already have an FPGA implementation going (albeit not complete.)

I looked at your [MC1](https://github.com/mbitsnbites/mc1) computer design and I am surprised that your are getting so much involved into video card. Of course, you do whatever your interests lead you to. Although I am wondering how important that is, since a serial terminal would do for most demo activity, and a fast network interface Ethernet or Fiber would seem to be more easily built and more powerful than a video card. 

But it does show that you think of yours as a personal computer more than a multi user time sharing computer. Which is cool. Along the lines that I am thinking too; why worry about multi-user if you can just have each user or application have his own computer? And so, there may be no need even for any UNIX like OS, as long as your favorite language runs on it. And that's the key, you want your programming language to be able to make use of all of the features you build. Doe GCC optimization do that? I would probably be happy if my computer could run Java (but using the optimizations you have), and then I could do anything I needed.

Your MC2 architecture reminds me of my old Commodore 64. CPU, memory bus used for ROM and RAM and the RAM simultaneously accessed by the Video subsystem, that also does the DRAM refresh. But thinking out of the box about the video card stuff: the days of VGA are numbered. And all new video interfaces seem to be backward compatible to the old way of doing things. Without the need to a high-speed scanning electron beam, why even worry about this? Why not just map an TFT or LED display panel onto memory? It appears that a TFT or display has some similarity to a DRAM chip, only that the transistor drives a LCD cell rather than a capacitor as it is in DRAM. So, therefore, it appears a massive overhead of serialization only to constantly transfer one array of charges in cells to another array of charges in cells. What would it be if we simply could treat a display panel like a direct memory element? The moment you change the pixel in that "memory" it would display. The refreshing of the display would be the exact same task as the refreshing of DRAM...

mbitsnbites, 2019-10-21:
Continuing.... W.r.t. ForwardCom, I think the two projects share similar ideas and motivations, though they may have started out slightly differently, and have made a few different design decisions (also I was not aware of neither ForwardCom nor RISC-V when I started MRISC32).

My next goal is a running (minimal) system on an FPGA. That's the motivation behind MC1.

There are several parts missing in the MRISC32 ISA to support a "proper" OS at this point (e.g. an exception model and memory management). Those parts can be added later (I have some ideas for how to do this, and the ISA is prepared for a few additions and extensions).

Once I have a running version of MRISC32 there are lots of interesting things to do. E.g. optimize the CPU architecture (or make a new one), add a complete exception model with efficient handling of large vector registers, improve the vector model (e.g. improved masking and register length handling), etc, etc.

I think that what would really help the MRISC32 project forward now is help with toolchains (GCC, LLVM, qemu, gdb, ...). I have very little experience in this area and often get stuck with annoying details.

I think that there is a place for MRISC32 - but much of that space is already being taken by RISC-V (which has a far stronger momentum). Unfortunately RISC-V is optimized for the ultra low end. I think that MRISC32 (and ForwardCom) has a better solution for higher end machines, and I also think that it's a simpler design (without the complexity of compressed instructions, and with vector operations and floating point integrated in the base ISA from the start instead of as separate extensions).

mbitsnbites, 2019-10-21:
Re: MC1 and video output. Yes, video is one of my interests, otherwise I would not have spent as much time on it (and yes, its inspired by Amiga, C=64 and Atari VCS, for instance).

I do think, however, that a direct video output is valuable for effective demos. You can easily get a feeling for the speed of vector operations, for instance, when you see the pixels animate @ 60 FPS. It's also more appealing as a showcase (I think that many YouTube videos of FPGA projects are quite dull when all you get is a blinking LED or a UART text dump).

mbitsnbites, 2019-10-21:
A few more comments:

I think that MRISC32 is more traditional than Fog's ForwardCom, which has gone to greater lengths to solve problems in contemporary CPU architectures. Where MRISC32 is a combination of innovations from the past (e.g. vector machines such as Cray-1 and RISC machines such as MIPS), ForwardCom goes a step further by innovating new concepts. IMO there are clear advantages with both approaches.

Before MRISC32 can be a viable architecture for general purpose computing (i.e. such that it could be compared to architectures like x86, Power and ARM in a meaningful way), it would have to be transformed into a 64-bit architecture (i.e. "MRISC64"). I currently regard MRISC32 as a proof of concept (or "template" if you will) for a future 64-bit ISA. While MRISC32 is already useful in its own right, one major deficiency is its lack of support for 64-bit IEEE floating point. Rather than making a contrived solution by storing 64-bit floating point values in 32-bit registers, I decided that it would be much cleaner to only support 32-bit floating point in a 32-bit architecture, and add full support for 64-bit floating point in a future 64-bit architecture instead. This is clearly a luxury that older 32-bit (or even 16-bit) ISA:s could not afford at the time when they were implemented, which is why floating point support in traditional ISA:s is often "bolted on" with a very different register model than the integer register model.

An MRISC64 ISA would be very similar to the MRISC32 ISA. Almost everything (the register model, instruction encoding, packed operations and vector operations, floating point support, etc, etc) would be almost identical. The only thing that may require a re-design (or extensions) would be how to handle 64-bit immediate values, and how to do sign extension and conversion between different word sizes (currently handled by the `SHUF` instruction, which is very 32-bit:ish).

gschadow, 2019-10-22:
What is the FPGA development kit you have been using? I am new to FPGA stuff, but I'd be interested in trying it out, just take the plunge. Do you intend to put the video part on the same FPGA? So essentially is the idea to create a single board single chip computer? Is that possible? If not, whould you have to design a board PCB solder all those nano component? OTOH, would all of this fit on a single stock FPGA dev kit?

mbitsnbites, 2019-10-23:
There are definitely use cases for doubles. E.g. when doing matrix inversions to solve equations (i.e. you have a complex computation involving both large and small numbers). The more important case, though, is to support existing software that uses double (even if it does not need it). For instance, the base numeric type in JavaScript is double. But I agree: most software can be written with single precision. It's commonly used in 3D graphics & physics due to the performance and storage space benefits.

A typical off-the-shelf FPGA board has a single FPGA chip plus additional components to support programming (USB/JTAG), I/O and debugging (LED:s, buttons, etc), plus some RAM (DRAM, SRAM, DDR3, ...). Since interfacing with external components (e.g. DRAM) is a source of problems and errors in any hardware design (timing issues, protocols, etc), I decided to rely completely on internal resources in the FPGA. A modern FPGA often has some fast, internal RAM, but not much (e.g. 50 - 300 KiB), but it's enough to run simple programs.

Here's an example board:

mbitsnbites, 2019-12-03:
@gschadow The first conclusive evidence of life from the MRISC32-A1 processor, running on an FPGA board:

https://vimeo.com/377042669
[image]

closed 2020-03-27 by mbitsnbites